android-rpi/device_brcm_rpi3 #16: CEC Support?

avenantsoftware, 2016-06-05:
https://source.android.com/devices/tv/HDMI-CEC.html

would it be as easy as adding the line:

frameworks/native/data/etc/android.hardware.hdmi.cec.xml:system/etc/permissions/android.hardware.hdmi.cec.xml

to:

PRODUCT_COPY_FILES := \

in:

https://github.com/peyo-hd/device_brcm_rpi3/blob/30e769330f6667bb1b25e79eab7e9af346a1e8c8/rpi3.mk ?

JoinTheRealms, 2016-07-14:
Anyone tried this?

ikalkov, 2016-08-10:
CEC [seems to need](https://source.android.com/devices/tv/HDMI-CEC.html#implementation) more than that:

> Device manufacturers should add the following line into PRODUCT_COPY_FILES in device.mk.

>PRODUCT_COPY_FILES += \
frameworks/native/data/etc/android.hardware.hdmi.cec.xml:system/etc/permissions/android.hardware.hdmi.cec.xml

>Depending on whether your device is a HDMI sink device or a HDMI source device, device manufacturers need to set ro.hdmi.device_type in device.mk for HdmiControlService to work correctly.

>For HDMI source devices, like Over the Top (OTT) boxes, set:
>PRODUCT_PROPERTY_OVERRIDES += ro.hdmi.device_type=4

>For HDMI sink devices, like panel TVs, set:
>PRODUCT_PROPERTY_OVERRIDES += ro.hdmi.device_type=0

maughanorama, 2016-10-22:
has anyone tried the above ^^^^^^ if we add the
 PRODUCT_PROPERTY_OVERRIDES += ro.hdmi.device_type=0
would CEC work on my samsung panel ?

ikalkov, 2016-10-22:
In my understanding, you need device_type=4, as the Raspberry Pi is a source of the HDMI signal and not it's sink. You are free to try it and post the result here.

maughanorama, 2016-10-22:
yer im doing the repo update...... going to take 6 hours apparently. didnt realize i needed the android source too :|

ikalkov, 2016-10-30:
@maughanorama Any update on this issue?

Alexandero89, 2016-12-12:
@maughanorama  any progress on this?
and if so, could you upload the image to any hoster? ;-)

ShapeShifter499, 2018-07-06:
is this dead?