android-rpi/device_brcm_rpi3 #17: SSL handshake issue

chriss745, 2016-06-06:
Hi Peter,

I got in contact with Sahaj in terms of an issue with ATV on rpi3, he suggested to open up an issue here with you.

I recently bought an rPi3 device, I installed an Android TV on it, it works perfectly, it runs smooth and I can sideload apps. Although when I installed Horizon Go app  (my local TV provider's web app for online TV streaming), I get a nasty SSL error:

06-05 16:04:16.554  3090  3095 I art     : Do partial code cache collection, code=55KB, data=62KB
06-05 16:04:16.556  3090  3095 I art     : After code cache collection, code=55KB, data=62KB
06-05 16:04:16.556  3090  3095 I art     : Increasing code cache capacity to 256KB
06-05 16:04:16.615  1990  2022 D AudioFlinger: mixer(0xb4183740) throttle end: throttle time(40)
06-05 16:04:16.653  3090  3090 W System.err: javax.net.ssl.SSLHandshakeException: Handshake failed
06-05 16:04:16.654  3090  3090 W System.err:    at com.android.org.conscrypt.OpenSSLSocketImpl.startHandshake(OpenSSLSocketImpl.java:429)
06-05 16:04:16.654  3090  3090 W System.err:    at android.net.SSLCertificateSocketFactory.verifyHostname(SSLCertificateSocketFactory.java:198)
06-05 16:04:16.654  3090  3090 W System.err:    at android.net.SSLCertificateSocketFactory.createSocket(SSLCertificateSocketFactory.java:443)
06-05 16:04:16.655  3090  3090 W System.err:    at org.apache.http.conn.ssl.SSLSocketFactory.createSocket(SSLSocketFactory.java:394)
06-05 16:04:16.656  3090  3090 W System.err:    at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:170)
06-05 16:04:16.656  3090  3090 W System.err:    at org.apache.http.impl.conn.AbstractPoolEntry.open(AbstractPoolEntry.java:169)
06-05 16:04:16.656  3090  3090 W System.err:    at org.apache.http.impl.conn.AbstractPooledConnAdapter.open(AbstractPooledConnAdapter.java:124)
06-05 16:04:16.657  3090  3090 W System.err:    at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:366)
06-05 16:04:16.657  3090  3090 W System.err:    at org.apache.http.impl.client.AbstractHttpClient.execute(AbstractHttpClient.java:560)
06-05 16:04:16.657  3090  3090 W System.err:    at org.apache.http.impl.client.AbstractHttpClient.execute(AbstractHttpClient.java:492)
06-05 16:04:16.657  3090  3090 W System.err:    at org.apache.http.impl.client.AbstractHttpClient.execute(AbstractHttpClient.java:470)
06-05 16:04:16.658  3090  3090 W System.err:    at android.net.http.AndroidHttpClient.execute(AndroidHttpClient.java:250)
06-05 16:04:16.658  3090  3090 W System.err:    at by.istin.android.xcore.source.impl.http.HttpAndroidDataSource.getInputSteam(SourceFile:258)
06-05 16:04:16.658  3090  3090 W System.err:    at by.istin.android.xcore.source.impl.http.HttpAndroidDataSource.getSource(SourceFile:302)
06-05 16:04:16.658  3090  3090 W System.err:    at by.istin.android.xcore.source.impl.http.HttpAndroidDataSource.getSource(SourceFile:54)
06-05 16:04:16.659  3090  3090 W System.err:    at by.istin.android.xcore.service.manager.AbstractRequestManager.execute(SourceFile:152)
06-05 16:04:16.659  3090  3090 W System.err:    at by.istin.android.xcore.service.manager.DefaultRequestManager.run(SourceFile:30)
06-05 16:04:16.659  3090  3090 W System.err:    at adz.run(SourceFile:67)
06-05 16:04:16.659  3090  3090 W System.err:    at adp.run(SourceFile:165)
06-05 16:04:16.660  3090  3090 W System.err:    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
06-05 16:04:16.660  3090  3090 W System.err:    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
06-05 16:04:16.660  3090  3090 W System.err:    at java.lang.Thread.run(Thread.java:761)
06-05 16:04:16.661  3090  3090 W System.err: Caused by: javax.net.ssl.SSLProtocolException: SSL handshake terminated: ssl=0x9cdcd7c0: Failure in SSL library, usually a protocol error
06-05 16:04:16.661  3090  3090 W System.err: error:10000410:SSL routines:OPENSSL_internal:SSLV3_ALERT_HANDSHAKE_FAILURE (external/boringssl/src/ssl/s3_pkt.c:610 0xadb27660:0x00000001)
06-05 16:04:16.661  3090  3090 W System.err: error:1000009a:SSL routines:OPENSSL_internal:HANDSHAKE_FAILURE_ON_CLIENT_HELLO (external/boringssl/src/ssl/s3_clnt.c:764 0xaa0eb736:0x00000000)
06-05 16:04:16.661  3090  3090 W System.err:    at com.android.org.conscrypt.NativeCrypto.SSL_do_handshake(Native Method)
06-05 16:04:16.662  3090  3090 W System.err:    at com.android.org.conscrypt.OpenSSLSocketImpl.startHandshake(OpenSSLSocketImpl.java:357)
06-05 16:04:16.662  3090  3090 W System.err:    ... 21 more
06-05 16:04:16.837  3090  3121 W EGL-DRI2: Native format mismatch: 0x1 != 0x5

I tried to fix it myself, but to be honest I have no clue where to start, https in general looks good in the build, if I install a Chrome, I am able to open up SSL based pages. The same build from Horizon Go works on many other devices in my house, so I believe the app is OK, just something maybe is missing from this Android build. I also tried another Android branch (RaspAnd) on my device, there the Horizon Go app works, but there are many issues with that Android build I would like to use your code on my device.

Please advise! Thank you!

Chris

peyo-hd, 2016-06-07:
rpi3 build is based on AOSP master branch with N-preview tag.
Crypto libraries could be differ from previous releases you tested.

Neither do I have detail understanding on this topic,
then need someone who has more experience.

peyo-hd, 2016-06-27:
It seem it's framework issue - not rpi3 porting issue.

closed 2016-06-27 by peyo-hd