android-rpi/device_brcm_rpi3 #20: Mesa 12.0 adaptation

peyo-hd, 2016-06-15:
Currently mesa 11.2 branch is being used for libGLES_mesa.so.
To use recent release of mesa, tried mesa 12.0 branch. 
Logcat does not show significant errors, but it's not showing graphics output.

Following 3 changes are applied for this trial.
(1) external_mesa3d : https://github.com/peyo-hd/external_mesa3d/tree/dmabuf
(2) external_drm_gralloc : https://github.com/peyo-hd/external_drm_gralloc/tree/dmabuf
(3) external_libdrm :  [0001-Add-new-drmGetNodeTypeFromFd-function.patch.zip](https://github.com/peyo-hd/device_brcm_rpi3/files/317183/0001-Add-new-drmGetNodeTypeFromFd-function.patch.zip)

There are some commits by Rob Herring & Emil Velikov - regarding dmabuf usage,
and currently no clear plan for me how to adapt those on rpi3 build.
Rob & Emil's commits could be seen on this link : https://github.com/peyo-hd/external_mesa3d/commits/dmabuf/src/egl/drivers/dri2

dengleiyy, 2016-08-03:
Dear author:
   My rpi could not be started, logcat cycle print :
"01-01 00:00:05.979   120   120 D libEGL  : loaded /system/lib/egl/libGLES_mesa.so
01-01 00:00:06.188   120   120 D GRALLOC-KMS: supported drm planes and formats
01-01 00:00:06.188   120   120 D GRALLOC-KMS: plane id 22
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format XR24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format AR24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format RA24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format RG16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format BG16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format AR15
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format XR15
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YU16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YV16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YU12
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YV12
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format NV12[    8.189892] random: nonblocking pool is initialized

01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format NV16
01-01 00:00:06.188   120   120 D GRALLOC-KMS: plane id 23
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format XR24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format AR24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format RA24
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format RG16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format BG16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format AR15
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format XR15
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YU16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YV16
01-01 00:00:06.188   120   120 D GRALLOC-KMS:     format YU12
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format YV12
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format NV12
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format NV16
01-01 00:00:06.189   120   120 D GRALLOC-KMS: plane id 24
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format XR24
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format AR24
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format RA24
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format RG16
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format BG16
01-01 00:00:06.189   120   120 D GRALLOC-KMS:     format AR15
```````"
Can you help me to analyze this problem?

Thanks.

peyo-hd, 2017-07-08:
Closing this since Mesa 17.0 applied to opv branch.

closed 2017-07-08 by peyo-hd