android-rpi/device_brcm_rpi3 #27: SchedPolicy: add_tid_to_group permission denied.

ShadowEO, 2016-06-24:
I see this happening a lot in logcat during many of the operations (attempting to pair a remote, etc)

Below is the logcat:
`06-24 20:20:44.496   431  1845 W SchedPolicy: add_tid_to_cgroup failed to write '2543' (Permission denied); fd=25
06-24 20:20:44.497   431  1845 W ActivityManager: Failed setting process group of 2543 to 0
06-24 20:20:44.497   431  1845 W System.err: java.lang.IllegalArgumentException
06-24 20:20:44.498   431  1845 W System.err: 	at android.os.Process.setProcessGroup(Native Method)
06-24 20:20:44.498   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.applyOomAdjLocked(ActivityManagerService.java:18762)
06-24 20:20:44.498   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.updateOomAdjLocked(ActivityManagerService.java:19246)
06-24 20:20:44.499   431  1845 W System.err: 	at com.android.server.am.BroadcastQueue.processCurBroadcastLocked(BroadcastQueue.java:262)
06-24 20:20:44.499   431  1845 W System.err: 	at com.android.server.am.BroadcastQueue.sendPendingBroadcastsLocked(BroadcastQueue.java:297)
06-24 20:20:44.499   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.sendPendingBroadcastsLocked(ActivityManagerService.java:16259)
06-24 20:20:44.500   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.attachApplicationLocked(ActivityManagerService.java:6205)
06-24 20:20:44.500   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.attachApplication(ActivityManagerService.java:6246)
06-24 20:20:44.500   431  1845 W System.err: 	at android.app.ActivityManagerNative.onTransact(ActivityManagerNative.java:507)
06-24 20:20:44.501   431  1845 W System.err: 	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2493)
06-24 20:20:44.501   431  1845 W System.err: 	at android.os.Binder.execTransact(Binder.java:453)
`

peyo-hd, 2016-06-27:
It seems this record is also a mistake.

ikalkov, 2016-07-11:
@peyo-hd Sorry for commenting on a closed issue, but did you solve this somehow?
I get a lot of those messages and it seems to be Android-related.

peyo-hd, 2016-07-12:
I did not check this issue on my build.
But let's re-open it, becasue it's observed on rpi3 build.

It seems kernel cgroup configuration is not matched to AOSP requirements.

ikalkov, 2016-07-12:
It looks like some commits from the android-kernel-common repository are missing in the RPi's kernel. Here is a bit of explanation: https://groups.google.com/forum/#!topic/android-kernel/TGRNgQgLXfI

Here is the common repository: https://android.googlesource.com/kernel/common/
I tried cherry-picking following commits, but no success for now:

6809864
1811046
f4adb71
69db8fc
703920c1 (leads to a minor conflict)

ikalkov, 2016-07-12:
OK, i just missed out one commit. It works now,
So here is what you need:

```
git checkout rpi3
git remote add aosp-common https://android.googlesource.com/kernel/common
git fetch aosp-common
git cherry-pick 6809864
git cherry-pick 1811046
git cherry-pick f4adb71
git cherry-pick 69db8fc
git cherry-pick 703920c
<resolve conflicts>
git cherry-pick 0d6687a
```

Or just use the patch file: [cgroups-patch.txt](https://github.com/peyo-hd/device_brcm_rpi3/files/359099/cgroups-patch.txt)

peyo-hd, 2016-07-13:
Uploaded your squashed patch in this commit : 
https://github.com/peyo-hd/kernel_rpi/commit/0fa37e4b44b5900f49af2fa709cd43bf593f9968

I thought all the commits collected by you are a type of re-base work of the original commit to 4.4 kernel.
https://android.googlesource.com/kernel/common/+/4a12178d3dd5b06a215076c60dc7cb124d711373

ikalkov, 2016-07-13:
Yeah, I didn't try this one as I thought it is outdated... A lot easier this way :D
Just tested it btw, works like a charm. I think you can close the ticket. Thanks!

ShadowEO, 2016-07-13:
Oh good, I actually did post this one in the correct place at least! Figured I'd attempt to report it, but think my logcat was a little too much..

closed 2016-07-23 by peyo-hd