android-rpi/device_brcm_rpi3 #29: No video on Panasonic Viera TV

geekpersonman, 2016-07-11:
Shows RPI test Rainbow square then just goes blank. Does connect to the network though

moonrise007, 2016-09-22:
Hello. I have Panasonic Viera TV with support up to 1080p 50/60 HZ. I see same behavior. Is there a way how to debug / check this problem ? It is working fine when I connect RPi to PC monitor. Thx

IceDragonUA, 2016-11-03:
problem can be solved by change
debug.drm.mode.force= "pixel_width x pixel_height"
in /system/build.prop file

for this you should install adb on you PC
than

adb connect ip_of_device
adb root
adb remount
adb pull /system/build.prop

change line for example to
debug.drm.mode.force="1280x720"

than

adb push build.prop /system/build.prop
adb reboot

caipivara, 2016-12-28:
@IceDragonUA this setting is set in `build.prop` without ""

I mean, its already like:
```
debug.drm.mode.force=1280x720
````

Not like:
````
debug.drm.mode.force="1280x720"
```

Tried both options but no success

ikalkov, 2017-02-26:
I'll close this issue as the solution was provided above. For other questions concerning video configuration please refer to the discussion group: https://groups.google.com/forum/#!topic/android-rpi/ciILTnABiKA

closed 2017-02-26 by ikalkov