android-rpi/device_brcm_rpi3 #32: Add support for official 7inch touch display

HoWilgh, 2016-07-31:
Can you please add support for the official 7 inch touch display?
This is needed to use the RPi with the display as tablet like system.

See the discussions like 
https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=122146&p=1017524#p1017524
or
https://groups.google.com/forum/#!topic/android-rpi/c5E6HjG7MHM
Tanks in advance,
BR,
Howil

Nvreformat, 2016-09-06:
I couldn't get it to work using the links you gave, is there something else you have to do to get it working?

dJos1991, 2016-10-19:
Any advance on this issue? I really like to use the touchscreen.

rosterloh, 2016-10-19:
I'm very interested in trying to implement this feature too

HoWilgh, 2016-10-27:
fabia...@gmail.com found some information about how to use the touchscreen on a custom kernel. Maybe this helps. Please see... 
https://github.com/raspberrypi/linux/issues/1147

anholt, 2016-11-07:
You're really waiting for https://github.com/anholt/linux/issues/8.  Until then, you may have some luck with merging in current rpi-4.4.y and using `vc4-fkms-v3d` instead of `vc4-kms-v3d` for the overlay (though you lose other modesetting features by doing so).

nikiwaibel, 2016-12-09:
it seems to work for me (display+touch), sort of, not perfect. i tried on android-7.1.0-r7 and modified the kernel compile options quite a bit:
> CONFIG_BACKLIGHT_CLASS_DEVICE=y
CONFIG_ARM_THUMBEE=y
CONFIG_BCM2835_SMI=y
CONFIG_TOUCHSCREEN_RPI_FT5406=y
CONFIG_BCM2835_DEVGPIOMEM=y
CONFIG_BCM2835_SMI_DEV=y
CONFIG_VT=y
CONFIG_CONSOLE_TRANSLATIONS=y
CONFIG_VT_CONSOLE=y
CONFIG_HW_CONSOLE=y
CONFIG_VT_HW_CONSOLE_BINDING=y
CONFIG_I2C_CHARDEV=y
CONFIG_I2C_MUX=y
CONFIG_I2C_MUX_PCA954x=y
CONFIG_SPI_BCM2835=y
CONFIG_SPI_BCM2835AUX=y
CONFIG_POWER_RESET_GPIO_RESTART=y
CONFIG_BCMA=y
CONFIG_BCMA_HOST_SOC=y
CONFIG_BCMA_DRIVER_GMAC_CMN=y
CONFIG_BCMA_DRIVER_GPIO=y
CONFIG_DRM_PANEL_SIMPLE=n
CONFIG_DRM_PANEL_RASPBERRYPI_TOUCHSCREEN=y
CONFIG_FIRMWARE_EDID=y
CONFIG_FB_FOREIGN_ENDIAN=y
CONFIG_FB_BOTH_ENDIAN=y
CONFIG_FB_BIG_ENDIAN=n
CONFIG_FB_LITTLE_ENDIAN=n
CONFIG_FB_BACKLIGHT=n
CONFIG_FB_BCM2708=y
CONFIG_FB_SSD1307=n
CONFIG_FB_RPISENSE=n
CONFIG_LCD_CLASS_DEVICE=y
CONFIG_LCD_PLATFORM=y
CONFIG_BACKLIGHT_RPI=y
CONFIG_BACKLIGHT_GPIO=y
CONFIG_DUMMY_CONSOLE=y
CONFIG_FRAMEBUFFER_CONSOLE=y
CONFIG_FRAMEBUFFER_CONSOLE_DETECT_PRIMARY=y
CONFIG_FRAMEBUFFER_CONSOLE_ROTATION=n
CONFIG_LOGO=y
CONFIG_LOGO_LINUX_MONO=y
CONFIG_LOGO_LINUX_VGA16=y
CONFIG_LOGO_LINUX_CLUT224=y
CONFIG_RTC_HCTOSYS=y
CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
CONFIG_SPEAKUP=n
CONFIG_STAGING_BOARD=y
CONFIG_FB_TFT=n
CONFIG_PWM=n
CONFIG_KDB_KEYBOARD=n
CONFIG_FONT_SUPPORT=y
CONFIG_FONTS=n
CONFIG_FONT_8x8=y
CONFIG_FONT_8x16=y

i am quite sure not all of those are required.

> ARCH=arm scripts/kconfig/merge_config.sh arch/arm/configs/bcm2709_defconfig android/configs/android-base.cfg android/configs/android-recommended.cfg ~/my.cfg

… is your friend, btw.

cunliffe, 2016-12-09:
Hmm, I tried to update /sys/devices/platform/soc/soc:gpu/graphics/fb0/virtual_size with 800x480 but the screen went blank.

As for the keyboard, navigation buttons not showing, change ro.sf.lcd_dentisty to 128 in the build.prop

nikiwaibel, 2016-12-09:
@cunliffe yes, screen goes blank (in my case it shows the text console - "A N D R O I D" is displayed) and as soon as you touch or swipe the touchscreen, the image appears without offset - at least in my case. but note all the other things i've turned on. maybe it works that way just because the virtual consoles are there in the "background". i also see quite some sort of little flicker when i swipe lists. it is not extraordinary smooth as usual.

keyboard: so close! i set it to 136 or sthg … will try 128. thanks!!!

nikiwaibel, 2016-12-09:
i get less flicker / less "not so responsive display", if framebuffer console is not compiled in. virtual console could stay there but i think it is quite useless without the framebuffer console.

nikiwaibel, 2016-12-10:
to work without external keyboard, i'd recommend to add LatinIME (and maybe OpenWnn) to PRODUCT_PACKAGES in device/brcm/rpi3/rpi3.mk. Terminal and SpareParts can be useful for development as well. a notificationbar may make sense also: device/brcm/rpi3/overlay/frameworks/base/packages/SystemUI/res/values/config.xml (change .tv.TvStatusBar to .tablet.TabletStatusBar). [somehow off-topic, i know]

ikalkov, 2016-12-12:
@nikiwaibel Thanks for that configuration. Unfortunately I face the same issues as @cunliffe. Android boots correctly, but the picture is shifted to the right and the touchscreen doesn't work at all :(

cunliffe, 2016-12-12:
@ikalkov, using your rtandroid build, if you add dtoverlay=rpi-ft5406 into the config.txt and add the rpi-ft5406.dtbo file in the overlays folder, the touchscreen will work.

Tried to build the kernel with @nikiwaibel config over the weekend, but for some reason it keeps crashing after the rainbow screen :(

nikiwaibel, 2016-12-12:
here is my config.txt:
===
hdmi_force_hotplug=0
#hdmi_drive=2
#config_hdmi_boost=4
#hdmi_group=2
#hdmi_mode=82
#disable_overscan=82
#framebuffer_width=1280
#framebuffer_height=720
kernel=zImage
device_tree=bcm2710-rpi-3-b.dtb
dtparam=audio=on
dtoverlay=vc4-kms-v3d,cma-256
initramfs ramdisk.img 0x01f00000
mask_gpu_interrupt0=0x400
avoid_warnings=2
gpu_mem=256
enable_uart=1
#dtoverlay=vc4-fkms-v3d,cma-64
#dtoverlay=vc4-kms-v3d,cma-256 (orig)
#dtoverlay=vc4-kms-v3d,cma-64
dtoverlay=i2c-rtc,ds3231
dtoverlay=rpi-backlight
dtoverlay=rpi-ft5406
===
@cunliffe : i've edited my post above and removed one of the kernel config lines. that caused a loop/crash in my case as well. the above works for me.

nikiwaibel, 2016-12-14:
let's see if android-7.1.1_r6 changed something (i've tried android-7.1.0_r7 above)

ikalkov, 2017-02-07:
I will close the issue as the required configuration options were posted above. Additionally, there is a patch file showing the minimal changes to make it work:

https://github.com/RTAndroid/android_vendor_brcm_rpi3_scripts/blob/aosp-7.1/video/raspberry-7inch-touchscreen

closed 2017-02-07 by ikalkov

dJos1991, 2017-02-13:
I followed @nikiwaibel steps to try the touchscreen working, but it's not working for me, see the pictures:
[image]
[image]

mohe2015, 2017-02-13:
I am not sure but I think I followed @ikalkov steps and it worked ( https://github.com/RTAndroid/android_vendor_brcm_rpi3_scripts/blob/aosp-7.1/video/raspberry-7inch-touchscreen ). But you also have to add all overlay files by copying them from kernel/rpi/arch/arm/boot/dts/overlays/*.dtbo to the overlays folder in the boot partition.

dJos1991, 2017-02-15:
Ok, now it is working with the changes from the link and adding all the overlay files to the boot partition.

The problem on the second image was that the screen was not fitting correct, but that is fixed now.
Is there something to do for fix the upside down problem?

mohe2015, 2017-02-15:
display_rotate=2 in config.txt. See http://elinux.org/RPiconfig#Video

nikiwaibel, 2017-02-17:
i am right now compiling again (android-7.1.1_r22). i really wonder how this can work out using _**only**_ the instructions
* https://github.com/peyo-hd/local_manifests/tree/nougat
* https://github.com/RTAndroid/android_vendor_brcm_rpi3_scripts/blob/aosp-7.1/video/raspberry-7inch-touchscreen
* https://github.com/peyo-hd/device_brcm_rpi3

because
* at no stage the kernel modules are build
* the kernel modules are never copied to the system image
* the (default) kernel config marks the touch code as module

i worked around by compiling the code directly into the kernel. i doubt having only the proper config.txt in place (but not the kernel modules or touch code itself) can make the touch working.

maybe i miss something some place. actually i'd love to go with kernel modules, but when and how do you put them to the system image?

mohe2015, 2017-02-17:
@nikiwaibel I'm not an expert with linux and android building but I think there are built here:
[image]

nikiwaibel, 2017-02-17:
* `make zImage` builds the kernel without modules
* `make dtbs` builds the device tree files

either `make modules` or `make` would compile the modules.

so i really wonder how anyone can get touch functionality. @mohe2015 do you have any .ko files (kernel objects - aka modules) within system.img? the modules are usually located in a lib/modules dir within /system, once the OS is running. further, a `lsmod` should display loaded modules.

mohe2015, 2017-02-17:
I can't check currently because I don't have a build ready but I don't think so because I strictly followed the build instructions.

dJos1991, 2017-02-17:
@mohe2015 That is not working. For Raspbian I use always lcd_rotate=2, but both are not working.

DId anyone tried to build de nougat-tablet branch and get it correctly working on the touch screen?

nikiwaibel, 2017-02-17:
@dJos1991: thx for the hint! i used to compile the nougat branch. now i've learned there is a nougat-tablet branch. will try that one now.

dJos1991, 2017-02-17:
@nikiwaibel Your Welcome ;).

I already compiled it and used the config from RTAndroid (https://github.com/RTAndroid/android_vendor_brcm_rpi3_scripts/blob/aosp-7.1/video/raspberry-7inch-touchscreen).
It is starting but the screen doesn't fit on the touch screen:
[image]

@mohe2015 Now i putted both display_rotate and lcd_rotate in the config file and now it shows the screen in the right direction.

nikiwaibel, 2017-02-17:
@dJos1991 try writing 800x480 to /sys/devices/platform/soc/soc:gpu/graphics/fb0/virtual_size - IIRC that blanked the screen in my case and then, after dragging something with the mouse, the screen offset was gone.

dJos1991, 2017-02-17:
I've putted 800x480 in that file, but still not good:
[image]

mohe2015, 2017-02-17:
Try removing the line 
`debug.drm.mode.force=1280x720`
 in device/brcm/rpi3/system.prop

ric96, 2017-02-17:
can you try the following on system.prop, or directly change it on build.prop
debug.drm.mode.force=800x480
ro.sf.lcd_density=160

dJos1991, 2017-02-18:
It is working now, after i put debug.drm.mode.force=800x480 in the build.prop, ro.sf.lcd_density=160 was already in it.
But i had to do the change in the sys/devices/platform/soc/soc:gpu/graphics/fb0/virtual_size file what @nikiwaibel posted, but is it possible to create a permanent solutions for that? I don't want to connect trough adb every time i start my rpi.

nikiwaibel, 2017-02-18:
@djos1991 can you commit all changes to your repo so I can reproduce this? Your touch works fine, right?

mohe2015, 2017-02-18:
A little bit offtopic: How can I switch to the nougat-tablet branch?

dJos1991, 2017-02-18:
@nikiwaibel I will commit al the changes later on tonight. Touch is working really fine!

@mohe2015 I forked the repo's to mine account, if you follow this steps you wil get the nougat-tablet branch automatically: [https://github.com/dJos1991/local_manifests](url). Else you can change .repo/local_manifests/default.xml line 12 like this before you do repo sync:
`  <project path="device/brcm/rpi3" name="device_brcm_rpi3" revision="nougat-tablet" remote="arpi"/>`

mohe2015, 2017-02-18:
@dJos1991 Thank you!
P.S. your link is broken. It shows the correct url but if you click on it it redirects to a wrong one.

dJos1991, 2017-02-18:
Thanks for reporting the broken link!
I think i commited al the changes now, I will try building it again to be sure that I'm not forget anything.

nikiwaibel, 2017-02-18:
@dJos1991 tried your branch, but no kernel modules, no nouch. display fine, except for the offset. mouse pointer sort of distorted.

dJos1991, 2017-02-18:
That is strange, I will check it out what the problem is.
I'm really new to Github and building Android, so please forgive me.

I changed the README pages for the instructions what I think I have done. 
If I enter lsmod on the terminal, it also doesn't show up any module.

I'm now syncing my repo and check everything and compare it with what i wrote on the readme pages.

nikiwaibel, 2017-02-18:
@dJos1991 it's all fine! you have no modules installed (`lsmod` won't show anything). your `rpitouch.cfg` file will compile the touch code directly into the kernel.

dJos1991, 2017-02-18:
ok fine that it is cleared out :). I didn't noticed your last edit before posting.

dJos1991, 2017-02-18:
@nikiwaibel haha no problem ;).

Yeah I think too that there are some extra things used in RTAndroid, but I don't have any idea what. 
Please keep me updated that i can try it out myself and add the information to my repo!

nikiwaibel, 2017-02-18:
on rtandroid:
* there is no screen offset. very nice, properly centered - cyanogenmod like - smooth boot video/animation!
* mousepointer disorted
* no touch modules either, so the kernel contains the touch code directly (touch works perfektly)
* it feels a bit more sluggish
* gallery and music app crash
* the task manager has dissortions in the app previews
* /system/build.prop sets ro.sf.lcd_display to 213. it should be 136. pixels are not square! width=800pixel/155mm → 131.1dpi / height=480pixel/86mm → 141.8dpi / average = 136.4dpi

nikiwaibel, 2017-02-19:
it seems to be quite a mess to keep branch nougat, nougat-tablet and now maybe nougat-rpitouch in sync! some audio stuff and the boot files have a newer version in nougat already.

ikalkov, 2017-02-19:
Hi guys. What you are suggesting was also my idea a couple of month ago. But @peyo-hd would like to keep this repository with a minimal configuration and let others fork and modify files as they need. This is actually no big deal for RTAndroid, as we have a pretty mature server architecture which takes care of keeping all the repositories in sync. This is basically why we decided to spit up the configuration and use separate scripts for setting up touchscreens. There are too many of them to provide a dedicated branch for every single one.

nikiwaibel, 2017-02-19:
It can be minimal, no problem. But I would keep all config, system.prop,
etc in the same branch. It will get a messy and outdated otherwise.

dJos1991, 2017-03-01:
@nikiwaibel Did you already find a solution for the screen offset?

nikiwaibel, 2017-03-01:
@dJos1991 unfort not yet. i managed to get the 4.4.50 kernel going (that's used by rtandroid), but the screen offset is still there. i also activated a custum boot animation (rtandroid has one as well), but that does not remove the offset either.

my next step is to reduce the number of patches required for 4.4.50 (github.com/raspberrypi/linux) to see what is really required to get it booted. then add specific kernel options releted to the framebuffer console and see if that makes any changes to the offset.

dJos1991, 2017-04-14:
I build today the lineageOS from @KonstaT on my raspberry pi 3 and did the changes for the touchscreen, but this give the same result.

nikiwaibel, 2017-04-16:
@dJos1991 yes. remove debug.drm.mode.force from device/brcm/rpi3/system.prop

sureshecer, 2017-05-23:
Hi all,

I'm new to Android world and trying to make my rpi 3 working on Android. So far I have build the images and flashed on sdcard. I'm having 7inch screen which works fine with official raspian image but when i'm flashing my builted android images nothing getting displayed.

Can anyone please help me out with clear instructions on how i can make rpi + 7 inch display working on Android.