android-rpi/device_brcm_rpi3 #33: Boot process stuck at android animation logo

sandeep-mvd, 2016-07-31:
Hi,

The boot process hangs at android logo. I waited for a long time but the home screen never comes up.
What could be possible going wrong? Need your suggestions.

I need ethernet working. Hence I am not able to use standard builds like https://github.com/Android-RPi3/android_releases_rpi3/releases

Thanks in advance.

Best Regards,
Sandeep

sandeep-mvd, 2016-08-01:
Is anyone else facing this issue or is it only me? Please respond.

peyo-hd, 2016-08-11:
For build troubleshoot, go to the forum : 
https://groups.google.com/forum/#!forum/android-rpi

closed 2016-08-11 by peyo-hd

jonathanisaac, 2018-02-26:
@binary-nerd did you solve it? I have the same problem and I had no response in the group.

binary-nerd, 2018-03-27:
@jonathanisaac I was able to get through it. But sorry, I cannot remember what did I do to resolve it. May be I had used an official build that included ethernet support or I disable a few modules here and there. Not really sure. I am sorry.