android-rpi/device_brcm_rpi3 #36: Again: Bluetooth Not working kernel 4.4.16-v7

shep2, 2016-08-19:
I saw this was closed under #13 but it seems to be broken again (and kernel was updated 2 weeks agao). Bluetooth option is shown but if you try enable , it won't enable.

tonyho, 2016-09-08:
Is there any method to downgrade the kernel version, so we can use the BT in RPI android 7 or 6?

danielcbit, 2016-09-27:
I'm too having problem with bluetooth, even in the rtandroid image the bluetooth is not working anymore.

If I try to enable it, after some time it gets enabled, but soon after start using it, the bluetooth dies.

I saw the following message in logcat:

01-01 00:06:00.767  1118  1147 I bt_core_module: module_start_up Starting module "controller_module"
01-01 00:06:01.410  1118  1137 E bt_hci  : command_timed_out hci layer timeout waiting for response to a command. opcode: 0xc03

Here is a log:
[bluetooth.txt](https://github.com/peyo-hd/device_brcm_rpi3/files/496068/bluetooth.txt)

ikalkov, 2016-10-30:
Can verify bluetooth is not working on RTAndroid 7.1_r4 with the latest kernel 4.4.23.

ikalkov, 2016-11-25:
@danielcbit Seems to be fixed now in RTAndroid.

c1100077, 2016-12-06:
I resolved this problem.

File -> /hardware/broadcom/libbt/src/hardware.c 
line -> about 235

static const fw_settlement_entry_t fw_settlement_table[] = {
    {"BCM43241", 200},
    {"BCM43341", 100},
    {"BCM43430A1", 1000}, // <- add this line
    {(const char *) NULL, 100}  // Giving the generic fw settlement delay setting.
};

After I add above line and rebuild system.img, Bluetooth worked.

ikalkov, 2016-12-07:
Sounds great, maybe @peyo-hd could integrate this change in the main repository?