android-rpi/device_brcm_rpi3 #41: Playing video is suck and screen resolution is only 720p

Slim1991, 2016-09-07:
Hi
I flashed your ROM and spent a night to test it.
- 720p video is good
- 1080p is very lag.
- Displaying colors of videos are very very bad, It looks totally different with sources.
And beside screen resolution is only 720p on my full HD TV.
If you don't have time to develop this ROM, guide me! I'm amateur but I want to help :D
Email me: slim.tbb@gmail.com

maughanorama, 2016-10-22:
i thinkn the issue is because there is no HW acceleration the display is experimental ( i believe, however i could be very wrong)

thienpow, 2016-10-27:
your video has sound?

ikalkov, 2017-02-26:
Please refer to the discussion group with questions concerning video config: https://groups.google.com/forum/#!topic/android-rpi/ciILTnABiKA

closed 2017-02-26 by ikalkov