android-rpi/device_brcm_rpi3 #44: Wiki Update to help build on devices with 8GB RAM [Solution Provided]

ric96, 2016-10-24:
On devices that have 8GB RAM the build fails due to GC crashing as less ram is allocated by jack.
the proper work around is by adding the following lines to buildconfig.mk

ANDROID_JACK_VM_ARGS := -Dfile.encoding=UTF-8 -XX:+TieredCompilation -Xmx4096m

the reason i am not creating a pull request is that this option sets a hard limit on memory allocation at 4096MB this means that if a device has more than 8GB ram say, 16 GB. It might suffer a slowdown as less ram is allocated.
So its better in wiki where people can modify if they have just 8GB of RAM.

thienpow, 2016-10-27:
you mean 8gb of disk space or ram? rpi3 come with 1gb ram, and this build come with a config to use 256mb as gpu_mem so leaving us having only 768mb ram.

ikalkov, 2016-10-30:
@ric96 You could update the README file (do you mean it with "wiki"?) and create a PR.

ikalkov, 2017-01-02:
Merged

closed 2017-01-02 by ikalkov