android-rpi/device_brcm_rpi3 #45: build error with external/mesa3d

wohenbushuang, 2016-10-30:
Hi, as I follow the steps in README with VirtualBox, ubuntu 16.06 and latest AOSP(android-7.1.0_r4). When I make it it encounter fails. I used `make -wd` and got the following debug info:

> including ./external/mesa3d/Android.mk ...
build/core/binary.mk:345: *** external/mesa3d/src/loader/Android.mk: libmesa_loader: LOCAL_CLANG is set to false.
Reaping losing child 0x250dfd0 PID 10092 
build/core/ninja.mk:168: recipe for target 'out/build-rpi3.ninja' failed
make: *** [out/build-rpi3.ninja] Error 1
Removing child 0x250dfd0 PID 10092 from chain.

I deleted the line 91 in `external/mesa3d/Android.mk`, then similar problem occurred on `gallium`, so I deleted line 101 again and it met errors when compiling (after including). So I had to undo the above deletion and deleted the line 28 in `external/mesa3d/Android.common.mk`:
> LOCAL_CLANG := false

This deletion works well so far and completes the including `.mk` part. I am still working on other compiling errors for my first image yet, so I'm not sure whether this deletion is proper.

thienpow, 2016-10-30:
i dont think u should delete anything...
try install the dependencies needed

sudo apt-get install git-core gnupg flex bison gperf build-essential   zip curl zlib1g-dev gcc-multilib g++-multilib libc6-dev-i386   lib32ncurses5-dev x11proto-core-dev libx11-dev lib32z-dev ccache   libgl1-mesa-dev libxml2-utils xsltproc unzip

wohenbushuang, 2016-10-30:
@thienpow I have done this, from http://source.android.com/source/initializing.html

ikalkov, 2016-10-30:
@wohenbushuang, you shouldn't be deleting anything. Just follow peyo's instructions (and close this issue on success). I think you forgot to clone his local manifest (mesa3d project gets replaced).

https://github.com/peyo-hd/local_manifests/blob/nougat/README

wohenbushuang, 2016-10-30:
@ikalkov actually I did clone this local_manifests....

ikalkov, 2016-10-30:
@wohenbushuang I have just tested the mesa3d repository - it compiles without errors.

ikalkov, 2017-01-07:
I close the issue as there were no more comments.

closed 2017-01-07 by ikalkov