android-rpi/device_brcm_rpi3 #48: google play inserted version?

halukmy, 2016-11-11:
please send google play enabled version 👍

ikalkov, 2016-11-15:
Including Google Play Services without paying licensing fees to Google is not legal.

nikiwaibel, 2016-12-14:
within https://git.embedded.rwth-aachen.de/rtandroid/downloads/raspberry-pi/ there is a script that adds the play service and it's app. it may be possible to develop a more general version of it …

ikalkov, 2016-12-14:
@nikiwaibel The script is designed to work with all distributions and will work especially with this one, as I used this repo originally for my work. I think he just doesn't like starting a script at all.

nikiwaibel, 2016-12-14:
@ikalkov i see.  @halukmy it is definitely not allowed to include the play store. only certified devices have the right to include it out of the box.

closed 2017-01-02 by ikalkov