android-rpi/device_brcm_rpi3 #49: Dialog is distorted

c1100077, 2016-12-06:
Hi guys

The dialog from the system is distorted on my device.

But the dialog from my application is not distorted.

I want to fix the distortion of the dialog from the system.

Is there anything you know about this problem?

ikalkov, 2016-12-07:
Could you attach a sceeenshot for a better understanding?

c1100077, 2016-12-08:
Hi ikalkov.

Screenshot is follow.

- [ ] From　system.
Execute code is here.
`Intent intent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);`
`startActivityForResult(intent, 1234);`

[image]

- [ ] From　my app.
Execute code is here.
`
AlertDialog.Builder alertDialog = new AlertDialog.Builder(MainActivity.this);
alertDialog.setTitle("Title");
alertDialog.setMessage("Visible？");
alertDialog.setPositiveButton("Yes", new DialogInterface.OnClickListener() {
    @Override
    public void onClick(DialogInterface dialog, int which) {
        textView.setText("Yes");
    }
});
alertDialog.setNegativeButton("No", new DialogInterface.OnClickListener() {
    @Override
    public void onClick(DialogInterface dialog, int which) {
        textView.setText("No");
    }
});
alertDialog.create().show();
`

[image]

nikiwaibel, 2016-12-09:
looks like the system thinks the screen is a couple of pixels wider :-/ seen this as well, but never tried to run a self compiled app.

cyberdesigner, 2017-03-06:
I get this too. This is not depending on self-compiled apps. even the system dialog of which launcher should be used shows this. the following messages seem to be related:

`03-06 16:41:49.377   415   426 I PackageManager: Setting last chosen activity com.android.tv.settings/.system.FallbackHome for user 0:
03-06 16:41:49.378   415   426 I PackageManager:   Action: "android.intent.action.MAIN"
03-06 16:41:49.378   415   426 I PackageManager:   Category: "android.intent.category.HOME"
03-06 16:41:49.378   415   426 I PackageManager:   Category: "android.intent.category.DEFAULT"
03-06 16:41:49.378   415   426 I PackageManager:   AutoVerify=false
03-06 16:41:49.382   415   582 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.HOME] flg=0x13000100 cmp=com.android.tv.settings/.system.FallbackHome} from uid 0 on display 0
03-06 16:41:49.384   415   582 W System  : ClassLoader referenced unknown path: 
03-06 16:41:49.385   415   582 W System  : ClassLoader referenced unknown path: /system/priv-app/TvSettings/lib/arm
03-06 16:41:49.390   415   582 D ApplicationLoaders: ignored Vulkan layer search path /system/priv-app/TvSettings/lib/arm:/system/priv-app/TvSettings/TvSettings.apk!/lib/armeabi-v7a:/system/lib:/vendor/lib for namespace 0xb31d8110
03-06 16:41:49.438   138   202 D AudioFlinger: mixer(0xa9b03000) throttle end: throttle time(30)
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (488, 605).
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (487, 605).
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (484, 607).
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (484, 610).
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (484, 612).
03-06 16:41:49.571   415   451 I InputDispatcher: Dropping event because there is no touchable window at (484, 613).
03-06 16:41:49.605   133  1604 D PermissionCache: checking android.permission.READ_FRAME_BUFFER for uid=1000 => granted (1133 us)
03-06 16:41:49.735   833   833 D FallbackHome: User unlocked and real home found; let's go!
03-06 16:41:49.776   415   769 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.HOME] flg=0x10000100 cmp=android/com.android.internal.app.ResolverActivity} from uid 0 on display 0
03-06 16:41:49.982   486   486 D skia    : --- SkAndroidCodec::NewFromStream returned null
03-06 16:41:49.988   486   486 D skia    : --- SkAndroidCodec::NewFromStream returned null
03-06 16:41:50.150   415   436 I ActivityManager: Displayed android/com.android.internal.app.ResolverActivity: +354ms
`

rtreffer, 2017-04-10:
I have the same issue with a beamer. What I can see is that the resolution is selected correctly on power on but as soon as the android splash screen takes over the resolution is reduced.

How can I pin userspace to keep the 1080p resolution?

rtreffer, 2017-04-23:
I'm wondering if this is related to https://github.com/android-rpi/device_brcm_rpi3/blob/nougat/system.prop#L1
```
debug.drm.mode.force=1280x720
```

Is there a reason why why the resolution is forced to 720p?

cyberdesigner, 2017-05-03:
@rtreffer I don't think your issue is related to this one. What you describe is a configuration problem. not the distortion of dialogs in the frame buffer overlay addressed in this issue.

nikiwaibel, 2017-05-05:
@rtreffer i also think it is not related. but, i see a similar issue with the mouse pointer.

fcatrin, 2017-07-09:
That's not a standard dialog, it is the Intent Resolver dialog that asks you to choose one Activity to use for the Intent. I tried to change the theme settings for that without success, later I found that the bug is only present in the software rendering path.  I finally enabled hardware rendering for that dialog only and it fixed the issue. I know that is only a workaround but it'll help.

```patch
--- ./frameworks/base/core/java/com/android/internal/app/ResolverActivity.java.original	2017-07-07 01:22:46.637354348 -0400
+++ ./frameworks/base/core/java/com/android/internal/app/ResolverActivity.java	2017-07-07 14:40:13.004965043 -0400
@@ -27,6 +27,7 @@
 import android.text.TextUtils;
 import android.util.Slog;
 import android.view.View.OnAttachStateChangeListener;
+import android.view.WindowManager;
 import android.widget.AbsListView;
 import com.android.internal.R;
 import com.android.internal.content.PackageMonitor;
@@ -202,6 +203,10 @@
     protected void onCreate(Bundle savedInstanceState, Intent intent,
             CharSequence title, int defaultTitleRes, Intent[] initialIntents,
             List<ResolveInfo> rList, boolean alwaysUseOption) {
+
+        getWindow().setFlags(WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED, 
+            WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED);
+
         setTheme(R.style.Theme_DeviceDefault_Resolver);
         super.onCreate(savedInstanceState);
 ```

maurossi, 2020-07-19:
Hello,
I found that Software Rendering path has issues that go beyong the ResolverActivity dialogue,
which is visible also in Chicken Invaders app.

The problem seems related to Software Rendering path not considering the correct stride being returned by gralloc at image buffer allocation.

Does someone have some clues on where the problem may be in Skia?
Thanks for any info

Mauro