android-rpi/device_brcm_rpi3 #50: Display problem during Android TV 7.1 installation on Raspberry Pi 3

JohnM87, 2016-12-11:
Hello !
I have a Raspberry pi 3, it works great with Recalbox, but when I want to install Android Tv, I have black screen.
I installed the image of Android TV 7.1 on the SD card and I also tried with an image of an older version of Android and I always the same result ... The multicolored square is displayed In 720p then the display changes format (unidentified). Then I have the Android logo white one second, the display format changes again and then nothing (black screen).
The orange LED on the Raspberry flashes, so I think it happens stuff, but no way to have the image ...
The manipulations on the Config.txt file do not change anything (10 combinations of different settings tried).
HELP ME PLEASE!

Nimothenicefish, 2016-12-12:
I have exactly the same problem.
Testé sur un écran Sony 65 "en HDMI et sur écran LCD 17" en DVI

Thanks for the help !

Nimothenicefish, 2016-12-12:
@JohnM87 add labels **bug** and **help wanted** and assign this issue to @peyo-hd 

Thank :wink:

hsstan, 2018-05-01:
I've found a solution.

In build.prop you have to comment the line with debug.drm.mode.force

gordomo, 2018-06-06:
@hsstan yes sir! that make my day! thanks a lot

closed 2018-06-06 by ikalkov