android-rpi/device_brcm_rpi3 #51: Setting app would stop working when click an item

tonyho, 2016-12-13:
After enter the settings context, click a random item would result in "Settings has stopped".
I use the newest branch as the guide, and patched the framework as the guide instruction.

Below is the relative logs:
`01-01 00:13:27.489   146   146 E GRALLOC-KMS: failed to perform page flip for primary (Device or resource busy) (crtc 43 fb 53))
01-01 00:13:27.490   146   146 E FramebufferSurface: error posting framebuffer: -16
01-01 00:13:29.564   433   763 I ActivityManager: START u0 {flg=0x8000 cmp=com.android.settings/.Settings$BluetoothSettingsActivity (has extras)} from uid 1000 on display 0
01-01 00:13:29.941  1318  1318 D AndroidRuntime: Shutting down VM
01-01 00:13:29.943  1318  1318 E AndroidRuntime: FATAL EXCEPTION: main
01-01 00:13:29.943  1318  1318 E AndroidRuntime: Process: com.android.settings, PID: 1318
01-01 00:13:29.943  1318  1318 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {com.android.settings/com.android.settings.Settings$BluetoothSettingsActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.app.ActionBar.setHomeAsUpIndicator(int)' on a null object reference
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3403)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3443)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2716)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.-wrap12(ActivityThread.java)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1461)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:102)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:154)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6088)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.app.ActionBar.setHomeAsUpIndicator(int)' on a null object reference
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at com.android.settingslib.drawer.SettingsDrawerActivity.showMenuIcon(SettingsDrawerActivity.java:228)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at com.android.settingslib.drawer.SettingsDrawerActivity.onResume(SettingsDrawerActivity.java:142)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1269)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.Activity.performResume(Activity.java:6783)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3380)
01-01 00:13:29.943  1318  1318 E AndroidRuntime: 	... 10 more
01-01 00:13:29.963   433   460 W ActivityManager:   Force finishing activity com.android.settings/.Settings$BluetoothSettingsActivity
01-01 00:13:30.050   433   460 W ActivityManager:   Force finishing activity com.android.settings/.Settings
01-01 00:13:30.208   433  1201 I OpenGLRenderer: Initialized EGL, version 1.4
01-01 00:13:30.209   433  1201 D OpenGLRenderer: Swap behavior 1
01-01 00:13:30.209   433  1201 W OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
01-01 00:13:30.209   433  1201 D OpenGLRenderer: Swap behavior 0
01-01 00:13:30.551   433   463 W ActivityManager: Activity pause timeout for ActivityRecord{6b0efbd u0 com.android.settings/.Settings$BluetoothSettingsActivity t8 f}
01-01 00:13:30.722  1142  1165 I OpenGLRenderer: Initialized EGL, version 1.4
01-01 00:13:30.722  1142  1165 D OpenGLRenderer: Swap behavior 1
01-01 00:13:30.723  1142  1165 W OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
01-01 00:13:30.723  1142  1165 D OpenGLRenderer: Swap behavior 0
01-01 00:13:31.596   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:31.617   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:31.701   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:31.717   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:31.734   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:31.934   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:32.001   433   464 D ViewRootImpl[settings]: updatePointerIcon called with position out of bounds
01-01 00:13:32.844   433   717 D ConnectivityService: NetworkAgentInfo [Ethernet () - 100] validation failed
01-01 00:13:41.534   433   463 W ActivityManager: Activity destroy timeout for ActivityRecord{6b0efbd u0 com.android.settings/.Settings$BluetoothSettingsActivity t8 f}
01-01 00:13:41.534   433   463 W ActivityManager: Activity destroy timeout for ActivityRecord{e6b9a54 u0 com.android.settings/.Settings t8 f}
01-01 00:13:41.535   433   463 I WindowManager: Destroying surface Surface(name=com.android.settings/com.android.settings.Settings) called by com.android.server.wm.WindowStateAnimator.destroySurface:2033 com.android.server.wm.WindowStateAnimator.destroySurfaceLocked:884 com.android.server.wm.WindowState.removeLocked:1471 com.android.server.wm.WindowManagerService.removeWindowInnerLocked:2487 com.android.server.wm.WindowManagerService.removeWindowLocked:2445 com.android.server.wm.WindowManagerService.removeWindowLocked:2314 com.android.server.wm.AppWindowToken.removeAllWindows:601 com.android.server.wm.AppWindowToken.removeAppFromTaskLocked:329 
`

As the log hint, maybe the framework patch guide should be updated now, which is not suitable for building the last code?

peyo-hd, 2017-01-22:
Try this : https://github.com/android-rpi/device_brcm_rpi3/wiki/Runtime-Errors#settings-crash

ric96, 2017-01-22:
I'll have a build out by Monday, its an updated tablet build and the settings app seems to be working fine without modifying the source.

ikalkov, 2017-02-01:
I close this, as solutions for both tablet and TV branches were proposed in the comments.

closed 2017-02-01 by ikalkov

wellcomez, 2018-02-17:
My problem is similar  but not only setting application any opengl app on system will crash  with similar reason .

I/OpenGLRenderer: Initialized EGL, version 1.4
D/OpenGLRenderer: Swap behavior 1
W/OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
D/OpenGLRenderer: Swap behavior 0
E/AndroidRuntime: FATAL EXCEPTION: GLThread 370
Process: com.brush.opengldemo, PID: 10473
java.lang.IllegalArgumentException: No configs match configSpec
at android.opengl.GLSurfaceView$BaseConfigChooser.chooseConfig(GLSurfaceView.java:877)
at android.opengl.GLSurfaceView$EglHelper.start(GLSurfaceView.java:1042)
at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1412)
at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1259)