android-rpi/device_brcm_rpi3 #61: merge various raspi configs [request for comments]

nikiwaibel, 2017-02-19:
i see more and more an issue regarding different raspi configs. thanks to @peyo-hd (!!!), we can get things going on various raspi3 platforms already.

right now, we have a nougat and a nougat-tablet branch here. @dJos1991 has created a nougat-rpi3touch branch within his repo. but some things in nougat-tablet (and nougat-rpi3touch) are outdated already (audio stuff) and i find it quite hard to merge. imagine we have 10 different raspi3 variations (i am sure there is more), do we really want to merge all branches every time nougat gets an update?

i suggest somehow to keep everything in one branch (nougat) and create different variations in `AndroidProducts.mk`. IMHO it should be something like:
```
PRODUCT_MAKEFILES := \
       $(LOCAL_DIR)/brcm_rpi3.mk
       $(LOCAL_DIR)/brcm_rpi3tablet.mk
       $(LOCAL_DIR)/brcm_rpi3rpitouch7.mk
```
maybe a bit more specific names.

dJos1991, 2017-02-19:
It is indeed hard to keep all the branches up to date. I'm very new to github, and I don't have a android knowledge, but I have a good knowledge of Linux in general. 

The reason that I forked it to my account was for trying things out to make it work with the 7" touch screen. 
@nikiwaibel, if u want, I can make you or someone else a contributer to my repo. 
I think that your solution is a good idea, but imho I think that it is better that we first make it working with the touchscreen.

nikiwaibel, 2017-02-19:
@dJos1991 you basically add the upstream (android-rpi) repo to your local git:
`git remote add upstream https://github.com/android-rpi/device_brcm_rpi3`
(verify by `git remote -v`)
then you can fetch the contents those contents by:
`git fetch upstream`
(veryfy by `git branch -a`)

no you can compare (diff) and/or merge
git checkout -B myBranch
git merge upstream/theirBranch

nikiwaibel, 2017-02-19:
nevertheless, 
* we have different config.txt and system.prop files
* maybe (in case kernel modules are not used) some different kernel configurations
* it looks like there are some changes in the overlay dir too

closed 2017-02-25 by nikiwaibel