android-rpi/device_brcm_rpi3 #68: recipe for target 'silentoldconfig' failed

guyansrg, 2017-03-17:
Hi,

I just try to compile kernel/rpi but I have a compile error.

$ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make zImage
scripts/kconfig/conf  --silentoldconfig Kconfig

*** Error during update of the configuration.

scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
Makefile:542: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
make: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/kernel.release'.

johnsonyao-EDT, 2017-10-24:
pi@raspberrypi:~/linux $ make
scripts/kconfig/conf  --silentoldconfig Kconfig

*** Error during update of the configuration.

scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
Makefile:546: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
make: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/kernel.release'.  Stop.

maykolo, 2021-04-20:
I also have this problem. How do you solve this problem?

wchi-123, 2022-08-21:
make mrproper # 用于删除所有的编译生成文件、内核配置文件(.config文件)和各种备份文件，所以几乎只在第一次执行内核编译前才用这条命令

make clean # 用于删除大多数的编译生成文件，但是会保留内核的配置文件.config，还有足够的编译支持来建立扩展模块。所以你若只想删除前一次编译过程的残留数据，只需执行make clean命令