android-rpi/device_brcm_rpi3 #69: Build failure

JeongyongShin, 2017-03-24:
after make command, below error occurred.
How can I do?

FAILED: /bin/bash -c "prebuilts/misc/linux-x86/bison/bison -d  --defines=out/host/linux-x86/obj/STATIC_LIBRARIES/libaidl-common_intermediates/aidl_language_y.h -o out/host/linux-x86/obj/STATIC_LIBRARIES/libaidl-common_intermediates/aidl_language_y.cpp system/tools/aidl/aidl_language_y.yy"
prebuilts/misc/linux-x86/bison/bison: m4 subprocess failed
ninja: build stopped: subcommand failed.
build/core/ninja.mk:148: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1

#### make failed to build some targets (42 seconds) ####

dJos1991, 2017-03-24:
Are you sure that you run this commands correct without errors?

`$ ARCH=arm scripts/kconfig/merge_config.sh arch/arm/configs/bcm2709_defconfig android/configs/android-base.cfg android/configs/android-recommended.cfg`
 `$ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make zImage`
` $ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make dtbs`

I think that you did something wrong because he is using the folder linux-x86 instead of arm.

JeongyongShin, 2017-03-25:
After run the below command
 $ ARCH=arm scripts/kconfig/merge_config.sh arch/arm/configs/bcm2709_defconfig android/configs/android-base.cfg android/configs/android-recommended.cfg 

below messages were shown..
How can I do?
Value requested for CONFIG_PM_RUNTIME not in final .config
Requested value:  CONFIG_PM_RUNTIME=y
Actual value:     

Value requested for CONFIG_PM_WAKELOCKS_LIMIT not in final .config
Requested value:  CONFIG_PM_WAKELOCKS_LIMIT=0
Actual value:     

Value requested for CONFIG_SUSPEND_TIME not in final .config
Requested value:  CONFIG_SUSPEND_TIME=y
Actual value:     

Value requested for CONFIG_MEMORY_STATE_TIME not in final .config
Requested value:  CONFIG_MEMORY_STATE_TIME=y
Actual value:

gw0, 2017-04-14:
I had the same problem. You are missing the `m4` command.

I had to install these requirements on Debian 8:

```
$ apt-get update && apt-get install build-essential gcc-arm-linux-gnueabihf python python-mako openjdk-8-jdk m4 libxml2-utils bc curl git vim-tiny unzip zip`
$ dpkg --add-architecture i386 && apt-get update && apt-get install libstdc++6:i386 libgcc1:i386 zlib1g:i386 libncurses5:i386
```