android-rpi/device_brcm_rpi3 #71: Ehernet UI in settings menu

AlejandroPerezRuiz, 2017-04-06:
I have a particular problem with Android 7 in rpi3. I have compiled Android 7 for my rpi3 and I only have the possibility to connect via ethernet, but I am using a switch witouth DHCP, so I need to put a static IP. However, there is no option in the settings to set a static IP for ethernet, is there any way to activate this option?

gw0, 2017-04-14:
As a workaround, you may connect over ADB and manually setup the static IP with `ip` or `ifconfig`. If you have DHCP, it automatically connects.

ikalkov, 2017-04-14:
@gw0, are you aware of the exact command format? I tried several approaches, but without a DHCP I can't ping in either direction. I have to admit, my experiences with networt setup are very limited.

nikiwaibel, 2017-04-14:
it seems the latest android x86 has no ethernet gui settings either. for JB and ICS there was a patch: https://github.com/gxben/aosp-ethernet

gw0, 2017-04-18:
@ikalkov Yes, now that I finally have a working build I can confirm that the standard Linux `ip` command is available. It works as it should, but Android 7 uses a more complex networking setup as usual, therefore one needs to specify the routing table to use. I connected over WiFi to get into ADB and used something commands like:

```
$ ip link set eth0 up
$ ip addr add 192.168.1.10/24 dev eth0
$ ip route add default via 192.168.1.1 dev eth0 table legacy_network
```

Where `192.168.1.10` is the IP address of RPi3 and `192.168.1.1` is your gateway (eg. router or forwarding server).

nikiwaibel, 2017-04-18:
do we really have no patch for 7.1 that integrates the ethernet settings in the Settings.apk!? i found
* https://github.com/gxben/aosp-ethernet (only for JB and ICS)
* http://elinux.org/images/9/98/Dive_Into_Android_Networking-_Adding_Ethernet_Connectivity.pdf

nikiwaibel, 2017-04-18:
do we really have no patch for 7.1 that integrates the ethernet settings in the Settings.apk!? i found
* https://github.com/gxben/aosp-ethernet (only for JB and ICS)
* http://elinux.org/images/9/98/Dive_Into_Android_Networking-_Adding_Ethernet_Connectivity.pdf
it seems there is no ethernet settings in AOSP, lineageos and android-x86 :(

gw0, 2017-04-18:
@ikalkov The above commands to manually configure an Ethernet interface (even without WiFi) work, but unfortunately DNS does not (browser says `net::ERR_INTERNET_DISCONNECTED`). To setup DNS the only command that worked for me:

```
$ ndc resolver setnetdns eth0 . 8.8.8.8 8.8.4.4
```

- <https://github.com/ukanth/afwall/wiki/DNS#resolver-commands>

So theoretically if you put this commands into `init.rc` it would setup Ethernet networking at boot time.

parinpatel, 2019-02-14:
@nikiwaibel Hi
Did you find any patch for adding  "Ethernet UI in the settings menu" [Android 7.1.1]?