android-rpi/device_brcm_rpi3 #73: Build failure at `out/target/product/rpi3/gen/EXECUTABLES/iw_intermediates/version.c`

gw0, 2017-04-14:
I followed the instructions but the main build process fails with:

```
$ make ramdisk systemimage
...
external/iptables/iptables/iptables-restore.c:46:55: warning: unused parameter 'version' [-Wunused-parameter]
static void print_usage(const char *name, const char *version)
                                                      ^
2 warnings generated.
[ 68% 20608/29939] target thumb C: iptables <= external/iptables/iptables/iptables.c
external/iptables/iptables/iptables.c:113:7: warning: missing field 'has_arg' initializer [-Wmissing-field-initializers]
        {NULL},
             ^
1 warning generated.
[ 68% 20641/29939] build out/target/product/rpi3/gen/EXECUTABLES/iw_intermediates/version.c
FAILED: /bin/bash -c "external/iw/version.sh out/target/product/rpi3/gen/EXECUTABLES/iw_intermediates/version.c"
fatal: No names found, cannot describe anything.
ninja: build stopped: subcommand failed.
build/core/ninja.mk:148: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1

#### make failed to build some targets (02:22:41 (hh:mm:ss)) ####
```

If I stop the jack server and rerun the build command, the same step still fails. As I understand the error message `fatal: No names found, cannot describe anything.` comes from `git describe --match=v*` in `external/iw/version.sh`. Any suggestions?

gw0, 2017-04-14:
Additional info:

```
$ cd ./external/iw
$ git describe --match=v*
fatal: No names found, cannot describe anything.
$ git describe
android-n-mr2-preview-2-3-g6a3e9a1
$ git status
Not currently on any branch.
nothing to commit, working tree clean
$ git checkout master
Previous HEAD position was 2af129e... Iw: Silence some warnings
Branch master set up to track remote branch master from aosp.
Switched to a new branch 'master'
$ git status
On branch master
Your branch is up-to-date with 'aosp/master'.
nothing to commit, working tree clean
$ git pull
Already up-to-date.
```

It seems that something changed 5 months ago in `version.sh` upstream: <https://android.googlesource.com/platform/external/iw/+/ed243cb4cbb6247bf4aeaed6c93b8167cce9b217>

As a workaround I removed the `./external/iw/.git` repository directory and now it continues.

ZephyrL, 2018-05-25:
I have also met up this error, and I searched for some solutions, some says to clean the git working tree before launch compiling is important, however this does not always work. Here's what I do to skip the issue:
`cd ($YOUR_AOSP_ROOT)/external/iw/version.sh`
comment all the commands from 
`if [-d .git] && ...` to the last `fi`,
except the only line 
`v="$version"`
in this case there is only one branch left and this bash file will never do git version check.
But I cannot promise that it does no side-effect.