android-rpi/device_brcm_rpi3 #76: Is it possible to support touch screen、sim card and camera？

andrewleo, 2017-05-07:


nikiwaibel, 2017-05-07:
touch already works, simcard depends very much on the manufacturer of the radio. camera should work as well - not sure if there is an "official" camera, as with the touch screen.

andrewleo, 2017-05-08:
Good news!
I wonder is there an "official" driver of touch screen？or is there official touch screen and the manual you can suggest to me ?

nikiwaibel, 2017-05-08:
https://www.raspberrypi.org/products/raspberry-pi-touch-display/
the driver is in the linux kernel.

andrewleo, 2017-05-08:
Sorry for my insufficient informations, *I want to install Android on my rpi3b.*
In this context, may touch screen, camera be supported?
This touch screen may only support Raspbian OS?

nikiwaibel, 2017-05-08:
i got it (otherwise you would not post in android-rpi ;-)). if you use the official touchdisplay (link above), then android is able to use it.

andrewleo, 2017-05-08:
Wonderful !  waveshare 7 inch touchscreen may be the official one?

nikiwaibel, 2017-05-08:
no idea if the waveshare 7" works.