android-rpi/device_brcm_rpi3 #81: Settings App crashes because of use of depreicated getActionBar

smndk, 2017-07-04:
Hi

I have successfully build a version of Android 7.1.2_r4 for RPI using this repository. But the Settings App crashes when I for example try to go the WiFi or Developer settings. Apparently the source code uses getActionBar several places, but it seems to be deprecated in Android 7. I have seen the patch in the Wiki, but it doesn't seem to be enough, and since the source is downloaded as part of the 7.1.2_r4 release, why should there be any need for such patches? Does anybody have any solution for this?

By the way the Music app crashes because another deprecated feature: MODE_WORLD_READABLE...

Kind regards,
Svend

peyo-hd, 2017-07-04:
https://github.com/android-rpi/device_brcm_rpi3/wiki/Runtime-Errors

smndk, 2017-07-05:
Thanks for the answer. I am sorry. I thought I had done those patches. 
But can you tell me, why it is necessary to patch code from the Android 7.1.2_r4 branch to make it work with the Android 7.1.2_r4 release?
The Music and Gallery Apps also use deprecated features that make these app crash.

closed 2017-08-21 by smndk