android-rpi/device_brcm_rpi3 #83: There is no BACK button when using mouse or pointer devices

fcatrin, 2017-07-09:
Many mouse or air mouse pointer devices use the RIGHT click as BACK, but this Android version interprets RIGHT as SECONDARY_BUTTON, which is right but makes the whole system unmanageable when using a pointer device only.

This optional patch will turn RIGHT click to be the BACK button.

```patch
--- ./frameworks/native/services/inputflinger/InputReader.cpp.original	2017-07-02 02:23:37.173738925 -0400
+++ ./frameworks/native/services/inputflinger/InputReader.cpp	2017-07-02 02:27:36.567681888 -0400
@@ -1193,9 +1193,9 @@
 
 void CursorButtonAccumulator::reset(InputDevice* device) {
     mBtnLeft = device->isKeyPressed(BTN_LEFT);
-    mBtnRight = device->isKeyPressed(BTN_RIGHT);
+    mBtnBack = device->isKeyPressed(BTN_RIGHT);
     mBtnMiddle = device->isKeyPressed(BTN_MIDDLE);
-    mBtnBack = device->isKeyPressed(BTN_BACK);
+    // mBtnBack = device->isKeyPressed(BTN_BACK);
     mBtnSide = device->isKeyPressed(BTN_SIDE);
     mBtnForward = device->isKeyPressed(BTN_FORWARD);
     mBtnExtra = device->isKeyPressed(BTN_EXTRA);
@@ -1220,7 +1220,7 @@
             mBtnLeft = rawEvent->value;
             break;
         case BTN_RIGHT:
-            mBtnRight = rawEvent->value;
+            mBtnBack = rawEvent->value;
             break;
         case BTN_MIDDLE:
             mBtnMiddle = rawEvent->value;
```

greystashh, 2018-04-27:
Would it be possible to write this as xposed module to enable this for other Android(TV) devices like nvidia shield too?

fcatrin, 2018-04-27:
I don't think so :-(

greystashh, 2018-04-28:
Do you see no chance to adapt this to ATV?

fcatrin, 2018-04-28:
I don't, that change is not isolated, you would need the source code from
those vendors to apply that patch
May be you can propose it to them so they can include it in their code.