android-rpi/device_brcm_rpi3 #87: should be root but can't change FS - device busy

grimace, 2017-08-17:
seems like I'm root running adb shell but when I try to change /system I get busy

rpi3:/ # mount -o remount,rw /system
mount: '/dev/block/mmcblk0p2'->'/system': Device or resource busy
1|rpi3:/ # 

Is there something I'm missing?

smndk, 2017-08-21:
The '/system' file system contains the Android system of the running OS. You cannot unmount or remount the '/system' file system of a running system. What are you trying achieve?

grimace, 2017-08-22:
I've done this on a number androids.  For instance, in order to edit 'build.prop'  and make a change,  you need to remount system as writeable.

mount -o remount, rw /system

allows me to do that.

smndk, 2017-08-22:
I am sorry - I stand corrected. Try the following:

mount -o rw,remount /system