android-rpi/device_brcm_rpi3 #89: RPI3 OTG port

smndk, 2017-08-21:
It there anyway to use the OTG USB port on the RPI3?

dJos1991, 2017-08-21:
Why would you use OTG USB? there are 4 normal USB ports on the RPI3.
The MicroUSB port is used to power the RPI3

nikiwaibel, 2017-08-22:
well: https://en.wikipedia.org/wiki/USB_On-The-Go

if you want to
* see the raspi sdcard over USB as a flash device (either for firmware/bootloader updates or
to access some part of the sdcard* access the raspi sdcard via USB as flash drive or MTP

smndk, 2017-08-22:
Hi

Thank you for your answers.
I would like to achieve two different things for two different purposes. One is making the OTG port a USB Host port just like the four dedicated host ports. I would in this case use the OTG port to connect "internal devices" keeping the four host ports free for "external devices".
The other purpose is to make the OTG port a USB HID device.

If this is possible what needs to be done? I suppose there needs to be made changes to the kernel?

I will in both cases supply power in other ways.

smndk, 2017-08-22:
OK. Reading a little bit more about the RPI3 I found out that the RPI3 OTG port cannot be used as a client device, as it is connected to an internal USB HUB.