android-rpi/device_brcm_rpi3 #90: Settings keeps stopping

yangjiel, 2017-08-29:
Hello there,

First of all, thank you for making Nougat Android support for RPI3.
I followed your instruction to compile and load the image into sd card. After the Android booted, the home screen is buggy, shows multiple mouses but the apps page, browser is good.

However, the main problem is I cannot use the setting, after I clicked any one of the options in the settings, it just crash and saying: Settings keeps stopping. There is no google pay that can be used. I want to install a terminal to see the boot log etc.

Is there anything wrong with my configuration?

ric96, 2017-08-29:
https://github.com/android-rpi/device_brcm_rpi3/wiki/Runtime-Errors

yangjiel, 2017-08-29:
Oh well, it seems I skipped a step applying patches. Thank you.

closed 2017-08-29 by yangjiel