android-rpi/device_brcm_rpi3 #95: Support of the official raspberry Pi camera on android-rpi

issamy, 2017-10-11:
I am currently using android-7.1.2_r19 on my raspberry pi 3. 
I would like to get working an official Rapsberry pi camera v1.3 or v2, connected through the CSI port of the raspberry pi. 
I installed a camera apk but when launching the application it crashes with error CameraService::Connect rejected (camera Hal module not loaded)

Is there anything to do to add support of official rpi camera on Android (modify the rpi3.mk, init file, buils some additional code - camera hal- ...)
If anyone has succeded, help would be much appreciated.

issamy, 2017-10-25:
Hasn't anyone used the camera using android-rpi on raspberry pi. 
If I add camera.default in rpi3.mk to add default camera hal to the compilation, I then have a crash:

01-01 00:06:20.684  3337  3337 I CameraService: CameraService process starting
01-01 00:06:20.689  3337  3337 I CameraService: Loaded "Default Camera HAL" camera module
01-01 00:06:20.690  3337  3337 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x0 in tid 3337 (cameraserver)
01-01 00:06:20.755  3342  3342 F DEBUG   : pid: 3337, tid: 3337, name: cameraserver  >>> /system/bin/cameraserver <<<
01-01 00:06:20.762  3342  3342 F DEBUG   :     #00 pc 0005fc0e  /system/lib/libcameraservice.so (_ZN7android12CameraModule31deriveCameraCharacteristicsKeysEjRNS_14CameraMetadataE+233)
01-01 00:06:20.762  3342  3342 F DEBUG   :     #01 pc 00060895  /system/lib/libcameraservice.so (_ZN7android12CameraModule13getCameraInfoEiP11camera_info+424)
01-01 00:06:20.762  3342  3342 F DEBUG   :     #02 pc 0005bb87  /system/lib/libcameraservice.so (_ZN7android16CameraFlashlight23createFlashlightControlERKNS_7String8E+146)
01-01 00:06:20.762  3342  3342 F DEBUG   :     #03 pc 0005bea5  /system/lib/libcameraservice.so (_ZN7android16CameraFlashlight14findFlashUnitsEv+96)
01-01 00:06:20.762  3342  3342 F DEBUG   :     #04 pc 0004ffa7  /system/lib/libcameraservice.so (_ZN7android13CameraService10onFirstRefEv+302)
01-01 00:06:20.762  3342  3342 F DEBUG   :     #05 pc 00000be3  /system/bin/cameraserver
01-01 00:06:20.762  3342  3342 F DEBUG   :     #06 pc 00000b25  /system/bin/cameraserver
01-01 00:06:20.763  3342  3342 F DEBUG   :     #08 pc 00000a18  /system/bin/cameraserver
01-01 00:06:26.230  3349  3349 I cameraserver: ServiceManager: 0xa8a19380
01-01 00:06:26.230  3349  3349 I CameraService: CameraService started (pid=3349)
01-01 00:06:26.230  3349  3349 I CameraService: CameraService process starting

Anyone can help ?
PS: camera is supported (at least USB camera which I tested) on android-things. From the trace I saw that the following camera hal is used: V4L2CameraHAL

Not sure if I should try to  to integrate this on android-rpi or debug the above crash using default camera hal

vinayd30, 2017-12-18:
I am also facing same error.
have you got any solution?

issamy, 2017-12-18:
Hi Vinay, 
Unfortunately, I have not been able to solve this "out of memory" error and I had to give up trying to get the camera working for now.
If you or someone else finds a solution, please share the solution for the community ;-)