android-rpi/device_brcm_rpi3 #123: Android OS for RPI4?

titanity, 2019-08-07:
Android OS for RPI4? Cant find it anywhere on the net...

saurabhin25, 2019-08-30:
Even i am searching

saurabhin25, 2019-08-30:
Android oR crankshaft still searching for rpi4

RaspberryPiFan, 2019-09-14:
I'm working on it: https://github.com/RaspberryPiFan/device_brcm_rpi4

saurabhin25, 2019-09-24:
Waiting for android 10 on my rpi4

RaspberryPiFan, 2019-10-12:
Build it!

RaspberryPiFan, 2019-10-12:
https://github.com/RaspberryPiFan/local_manifests

saurabhin25, 2019-10-12:
How to download

RaspberryPiFan, 2019-10-12:
RTFM!!!

titanity, 2019-10-12:
RaspberryPiFan i cant wait to see it. Do you have a finished image ready to burn to USB for the 4gb version?

RaspberryPiFan, 2019-10-13:
> RaspberryPiFan i cant wait to see it. Do you have a finished image ready to burn to USB for the 4gb version?

No, I'm sorry, I haven't build or tested this, but I'm going to do it soon. Do you need an totorial how to build it yourself?

RaspberryPiFan, 2019-10-13:
You can't build this on Windows or on Raspberry Pi, you need an PC with linux

RaspberryPiFan, 2019-10-13:
Run this on Debian or Ubuntu:
```
mkdir ~/bin
PATH=~/bin:$PATH
curl https://storage.googleapis.com/git-repo-downloads/repo > ~/bin/repo
chmod a+x ~/bin/repo
cd
mkdir android
cd android
repo init -u https://android.googlesource.com/platform/manifest -b android-10.0.0_r2
git clone https://github.com/android-rpi/local_manifests .repo/local_manifests -b android-10
repo sync
sudo apt install gcc-arm-linux-gnueabihf libssl-dev
cd kernel/rpi
ARCH=arm scripts/kconfig/merge_config.sh arch/arm/configs/bcm2711_defconfig kernel/configs/android-base.config kernel/configs/android-recommended.config
ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make zImage
ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make dtbs
sudo apt install python-mako
source build/envsetup.sh
lunch rpi4-eng
make ramdisk systemimage vendorimage
```

RaspberryPiFan, 2019-10-13:
Next step:

> # Prepare sd card
>  Partitions of the card should be set-up like followings.
>   p1 256MB for BOOT : Do fdisk : W95 FAT32(LBA) & Bootable, mkfs.vfat
>   p2 640MB for /system : Do fdisk, new primary partition
>   p3 128MB for /vendor : Do fdisk, new primary partition
>   p4 remainings for /data : Do fdisk, mkfs.ext4
>  Set volume label for /data partition as userdata
>   : use -L option of mkfs.ext4, e2label command, or -n option of mkfs.vfat

RaspberryPiFan, 2019-10-13:
```
cd out/target/product/rpi4
sudo dd if=system.img of=/dev/<p2> bs=1M
sudo dd if=vendor.img of=/dev/<p3> bs=1M
```
replace <p2> and <p3> with the partition

  device/brcm/rpi4/boot/* to p1:/
  kernel/rpi/arch/arm/boot/zImage to p1:/
  kernel/rpi/arch/arm/boot/dts/bcm2711-rpi-4-b.dtb to p1:/
  kernel/rpi/arch/arm/boot/dts/overlays/vc4-kms-v3d.dtbo to p1:/overlays/vc4-kms-v3d.dtbo
  out/target/product/rpi4/ramdisk.img to p1:/

saurabhin25, 2019-10-14:
Sorry i am a newbee cant do this...please make a downloadable file if u can ... Its been a month searching for android for my rpi4

RaspberryPiFan, 2019-10-14:
> Sorry i am a newbee cant do this...please make a downloadable file if u can ... Its been a month searching for android for my rpi4

Sorry, I can't do this at the moment, but eventually next week.

saurabhin25, 2019-10-14:
Okay ur name will be written with golden words

RaspberryPiFan, 2019-10-14:
@saurabhin25  Do you have an PC with windows 10?

RaspberryPiFan, 2019-10-14:
If yes, I can tell you how to build it.

RaspberryPiFan, 2019-10-14:
But you can't use your PC while building.
And I don't know if it works.

titanity, 2019-10-14:
Yes i do and i also got ubuntu and win10 together as multiboot

saurabhin25, 2019-10-14:
Yes i have windws 10

titanity, 2019-10-15:
when i run :
git clone https://github.com/android-rpi/local_manifests .repo/local_manifests -b android-10
in Ubuntu it says :
Remote branch android-10 not found in upstream origin

titanity, 2019-10-15:
[https://ibb.co/nz3bN8M](url)

titanity, 2019-10-15:
https://ibb.co/nz3bN8M

Berisan, 2019-10-15:
Try git clone https://github.com/android-rpi/local_manifests .repo/local_manifests -b android-10

RaspberryPiFan, 2019-10-15:
You're doing it wrong.

saurabhin25, 2019-10-15:
I have windows 10 pc please help me

RaspberryPiFan, 2019-10-15:
Yes, do you have the latest win10 version?

saurabhin25, 2019-10-15:
Yes

RaspberryPiFan, 2019-10-15:
Open PowerShell as administrator

RaspberryPiFan, 2019-10-15:
Type `Enable-WindowsOptionalFeature -Online -FeatureName Microsoft-Windows-Subsystem-Linux`

RaspberryPiFan, 2019-10-15:
Reboot your PC and install this: https://www.microsoft.com/store/apps/9MSVKQC78PK6

RaspberryPiFan, 2019-10-15:
Tell me when you're finished.

RaspberryPiFan, 2019-10-15:
@saurabhin25  Did it work?

RaspberryPiFan, 2019-10-15:
@saurabhin25 ?

Berisan, 2019-10-15:
@RaspberryPiFan the branch name was changed. https://github.com/android-rpi/local_manifests/commit/67d814a03af6bd86d96ec5aa509d6a52d3fcc5af

RaspberryPiFan, 2019-10-15:
It was a typo, I wanted to use my repository.

saurabhin25, 2019-10-15:
Please make a downloadable file please

titanity, 2019-10-15:
RaspberryPiFan do you know how many people waiting for android for RPI4 atm? You are really gonna be famous if you can make a working android for RPI4... Specially if its android 10. Im on discord with ppl who is quite famous on youtube who didnt even finish a image yet

saurabhin25, 2019-10-15:
Ur name will be written with golden words

RaspberryPiFan, 2019-10-15:
I don't know if it works, but if someone wants to build it yourself, I can tell him how.

titanity, 2019-10-15:
You use teamspeak? i got my own private server if so... im building atm...

i had a stop when repo sync .. it said:

Host key verification failed.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

titanity, 2019-10-15:
this message i got after 10 mins runtime

RaspberryPiFan, 2019-10-15:
You can contact me at discord https://discord.gg/AQcrHbD

saurabhin25, 2019-10-16:
Can i build this on buster

titanity, 2019-10-16:
No you need a pc with linux

titanity, 2019-10-16:
@RaspberryPiFan Let me know if you get the time today. Because im done with the first part. But im stuck at compiling the kernel

RaspberryPiFan, 2019-10-16:
git clone https://github.com/raspberrypi/tools ~/tools
echo PATH=$PATH:~/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin >> ~/.bashrc
source ~/.bashrc
sudo apt-get install git bison flex libssl-dev

RaspberryPiFan, 2019-10-16:
This will take some time...

RaspberryPiFan, 2019-10-17:
Further comments: https://github.com/RaspberryPiFan/device_brcm_rpi4/issues/1

botswanabub, 2019-10-18:
received _this_

`  HOSTCC  scripts/basic/fixdep
scripts/basic/fixdep.c:97:10: fatal error: sys/types.h: Datei oder Verzeichnis nicht gefunden
 #include <sys/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.host:90: recipe for target 'scripts/basic/fixdep' failed
make[1]: *** [scripts/basic/fixdep] Error 1
Makefile:467: recipe for target 'scripts_basic' failed
make: *** [scripts_basic] Error 2
`
_while_ 

`ARCH=arm scripts/kconfig/merge_config.sh arch/arm/configs/bcm2711_defconfig kernel/configs/android-base.config kernel/configs/android-recommended.config`

saurabhin25, 2019-10-18:
@titanity done?

RaspberryPiFan, 2019-10-18:
@botswanabub Du kannst mit mir Deutsch sprechen.

RaspberryPiFan, 2019-10-18:
Ich arbeite am Problem.

RaspberryPiFan, 2019-10-18:
@titanity  The conection doesn't works

botswanabub, 2019-10-19:
@RaspberryPiFan auch gut, aber vielleicht haben ja andere auch das Problem.
Leider bin ich auf dem Feld absolut unerfahren, daher kann ich da nicht wirklich mit helfen.

saurabhin25, 2019-10-19:
Waiting for the release

RaspberryPiFan, 2019-10-19:
@botswanabub Ich konnte dein Problem nicht reproduzieren. Ich arbeite gemeinsam mit @titanity am Build, dabei ist der Fehler nicht aufgetreten. Warte am besten auf die fertige Version, wahrscheinlich erscheint sie schon morgen.

botswanabub, 2019-10-19:
wenn ich dir mit ner Info helfen kann, sage bescheid. Ansonst vielleicht noch:

nach dem SYNC scheitert es bei mir grundsätzlich schon daran, dass er das Verzeichnis gar nicht geladen hat: `cd kernel/rpi`

[44 more replies]

closed 2020-04-19 by peyo-hd