android-rpi/device_brcm_rpi3 #130: System Trace and Settings

nicodt, 2020-05-11:
I am fairly new to android compiles but have worked with numerous kernels and my background is C/C++.
I patched and compiled quite easily without problems , and flashed the sd card it boots but only runs settings and system trace .So it seems the is the android UI or default system is not running. When I do click on settings it brings up the settings tab but only highlights where the mouse cursor is. I have a mouse plugged in at moment and the course looks funky. Thanks in advance for any comments good or bad.

sf0p, 2020-06-13:
same as i

moukail, 2020-06-23:
me too, i get this errors from dmesg.

```
[    0.711824] [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
[    0.711872] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    0.711987] vc4-drm soc:gpu: master bind failed: -517
[    3.172841] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    3.180613] vc4-drm soc:gpu: master bind failed: -517
[   46.217234] init: Could not find 'android.hardware.graphics.mapper@3.0::IMapper/default' for ctl.interface_start
[   46.253493] [drm:vc4_bo_create] *ERROR* Failed to allocate from CMA:
[   46.260796] [drm]                            V3D: 152632kb BOs (199)
[   46.268301] [drm]                     V3D shader:     40kb BOs (10)
[   46.274858] [drm]                           dumb:   3600kb BOs (1)
[   46.281267] [drm]                         binner:  16384kb BOs (1)
[   46.287623] [drm]                            BCL:      4kb BOs (1)
[   46.294486] [drm]                kernel BO cache:     12kb BOs (2)
[   46.807506] init: Received control message 'interface_start' for 'android.hardware.graphics.mapper@3.0::IMapper/default' from pid: 125 (/system/bin/hwservicemanager)
[   46.824068] init: Could not find 'android.hardware.graphics.mapper@3.0::IMapper/default' for ctl.interface_start
```

zjwhonor, 2020-09-01:
me too,how to fix this issue?

zjwhonor, 2020-09-01:
It's raspberrypi laucher, please refer to this link.
https://github.com/android-rpi/device_arpi_rpi4/issues/24