android-rpi/device_brcm_rpi3 #131: Vulkan driver for rpi3?

alepi, 2020-06-25:
Hi all, first of all thank you for making this. I tried it on a pi4 and apart from a few glitches it works beautifully.

I was wondering if you were interested in making a build for rpi3 that includes the vulkan driver that has been recently created (https://github.com/Yours3lf/rpi-vk-driver).

Let me know if you are interested and if you would like any help with it.

peyo-hd, 2020-06-26:
Hi,  thanks for info.

Nowadays, I spend times only for rpi4 build (https://github.com/android-rpi/device_arpi_rpi4).
For Vulkan, I'm just waiting for rpi4 mesa support be mature (https://www.raspberrypi.org/blog/vulkan-update-now-with-added-source-code)

Keeping the post for other's reference.

alepi, 2020-06-26:
ok thanks anyway :-)

closed 2020-06-26 by alepi