android-rpi/device_brcm_rpi3 #138: Not able to compile the kernel code showing 'selected processor does not support cpsid i in arm mode

iqsundar, 2023-02-07:
Hi Team,
when i try to compile the kernel code zImage, the following errors are showing. i followed the steps as explained in the git repo, but still showing error. i tried to compile it for raspberry PI 3+ hardware. ASOP build is fine.

$ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make  zImage
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      init/main.o
/tmp/ccpjcDEd.s: Assembler messages:
/tmp/ccpjcDEd.s:2106: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccpjcDEd.s:2248: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccpjcDEd.s:2308: Error: selected processor does not support `cpsie i' in ARM mode
/tmp/ccpjcDEd.s:2640: Error: selected processor does not support `cpsie i' in ARM mode
/tmp/ccpjcDEd.s:2723: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccpjcDEd.s:2734: Error: selected processor does not support `dmb ishst' in ARM mode
/tmp/ccpjcDEd.s:2743: Error: selected processor does not support `dmb ishst' in ARM mode
make[1]: *** [scripts/Makefile.build:334: init/main.o] Error 1
make: *** [Makefile:1115: init] Error 2

dzc1234ok, 2023-02-08:
Hi bro,
I met the same issues, using this cross compiler instead. 
https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64.

iqsundar, 2023-02-10:
@dzc1234ok ,
thank you very much bro.. 
it is compiled and working fine. thanks

closed 2023-02-10 by iqsundar