android-rpi/device_brcm_rpi3 #140: kernel duplicate definition compile error

mihaimacarie98, 2023-04-19:
`
xxx:/mnt/sdb1/aosp-raspi3/kernel/rpi$ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make zImage                                                                        scripts/kconfig/conf  --syncconfig Kconfig
  CALL    scripts/checksyscalls.sh
  HOSTLD  scripts/dtc/dtc
/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:99: scripts/dtc/dtc] Error 1
make[1]: *** [scripts/Makefile.build:638: scripts/dtc] Error 2
make: *** [Makefile:1130: scripts] Error 2
`

mihaimacarie98, 2023-04-19:
Fixed that but now I get this:
```

  CC      kernel/bounds.s
  CC      arch/arm/kernel/asm-offsets.s
  CALL    scripts/checksyscalls.sh
  CC      scripts/mod/empty.o
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/compile.h
  CC      init/main.o
/tmp/cc6JuMj4.s: Assembler messages:
/tmp/cc6JuMj4.s:2106: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/cc6JuMj4.s:2248: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/cc6JuMj4.s:2308: Error: selected processor does not support `cpsie i' in ARM mode
/tmp/cc6JuMj4.s:2640: Error: selected processor does not support `cpsie i' in ARM mode
/tmp/cc6JuMj4.s:2723: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/cc6JuMj4.s:2734: Error: selected processor does not support `dmb ishst' in ARM mode
/tmp/cc6JuMj4.s:2743: Error: selected processor does not support `dmb ishst' in ARM mode
make[1]: *** [scripts/Makefile.build:334: init/main.o] Error 1
make: *** [Makefile:1116: init] Error 2
```

reuelqueiroz, 2024-04-04:
I think you need to verify if you have linaro toolchain installed correctly

Download [https://releases.linaro.org/components/toolchain/binaries/7.4-2019.02/arm-linux-gnueabihf/ gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf.tar.xz](https://releases.linaro.org/components/toolchain/binaries/7.4-2019.02/arm-linux-gnueabihf/gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf.tar.xz)

I had the same problem because I didn't install linaro toolchain correctly but when I fixed it, it started working well